SAtacker/bb-config #32: Overlay File System

SAtacker, 2021-08-10:
I don't think it's possible to programmatically do it once the system is started. 
The workaround could be to use system calls.

ArthurSonzogni, 2021-08-10:
I am not sure what is the overlay file system. Maybe @VedantParanjape or @lorforlinux can help here.

VedantParanjape, 2021-08-10:
> I am not sure what is the overlay file system. Maybe @VedantParanjape or @lorforlinux can help here.

Hi Arthur,
To give a high level understanding, overlays are files which describe to the kernel the address of peripherals/subsystems inside a processor. These are loaded during boot and cannot be changed dynamically. @lorforlinux is expert, he can add more detail to my comment.

For a more detailed understanding read this: https://octavosystems.com/app_notes/osd335x-design-tutorial/osd335x-lesson-2-minimal-linux-boot/linux-device-tree-overlay/

SAtacker, 2021-08-11:
@VedantParanjape https://www.kernel.org/doc/html/latest/filesystems/overlayfs.html this is what I meant by OverlayFS
My proposal https://elinux.org/BeagleBoard/GSoC/2021_Proposal/beagle_config#Overlay_File_System highlights why it could be useful.

lorforlinux, 2021-08-11:
@VedantParanjape The Overlay file system is different from the Device Tree Overlay, Please confirm @SAtacker.

SAtacker, 2021-08-11:
> @VedantParanjape The Overlay file system is different from the Device Tree Overlay, Please confirm @SAtacker.
> 

Yes, it's different from dt-overlays

VedantParanjape, 2021-08-14:
> @VedantParanjape https://www.kernel.org/doc/html/latest/filesystems/overlayfs.html this is what I meant by OverlayFS
> My proposal https://elinux.org/BeagleBoard/GSoC/2021_Proposal/beagle_config#Overlay_File_System highlights why it could be useful.

TIL, name is a bit misleading.

SAtacker, 2021-08-14:
I have added a request on forum for this functionality to be included in debian images for beaglebone boards.
https://forum.beagleboard.org/t/protecting-the-sd-card-from-write-corruption-using-overlayfs/30543

RobertCNelson, 2021-08-16:
@SAtacker can you add an "enablement"..

append "overlayroot=tmpfs" to cmdline in /boot/uEnv.txt  all you need is overlayroot:

```
debian@BeagleBone:~$ dpkg --list | grep overlayroot
ii  overlayroot                               0.47ubuntu1rcnee0~bullseye+20210815     all          use an overlayfs on top of a read-only root filesystem
```

Regards,

SAtacker, 2021-08-16:
Thanks for this. I'll try to implement ut in beaglecfg.

RobertCNelson, 2021-08-16:
The default is:

```
overlayroot=tmpfs
```

But uses can "manually tweak" tweak this..

Keep Swap in /etc/fstab enabled...
```
overlayroot=tmpfs:swap=1
```

THe rest is:

```
debian@BeagleBone:~$ cat /etc/overlayroot.conf 
# This is the overlayroot config file
# By default, overlayroot is not enabled.
# To enable overlayroot:
#   1) edit the 'overlayroot' definition below
#   2) reboot
#
# Supported values:
#  * overlayroot=tmpfs or overlayroot=tmpfs:PARAMETERS
#    write all changes to a temporary (ram only) backing device
#    A tmpfs mount will be created, and usable filesystem can
#    grow to 1/2 available memory.
#
#    available parameters:
#     * see COMMON PARAMETERS
#
#    examples:
#     overlayroot=tmpfs
#     overlayroot=tmpfs:swap=1
#
#  * overlayroot=DEVICE or overlayroot=device:PARAMETERS
#    mount DEVICE as overlayfs and write changes there
#    device must already have kernel mountalbe filesystem on it.
#
#    available parameters are:
#     * dev: default: "" [REQUIRED]
#       use given device for backing filesystem.
#       Note, 'overlayroot=/dev/vdb' is translated to
#             'overlayrooot=device:dev=/dev/vdb'
#     * timeout: default: 0
#       if 'dev' provided does not exist, wait up to many seconds for
#       it to appear.
#     * see COMMON PARAMETERS
#
#    examples:
#      overlayroot=/dev/xvdb
#      overlayroot=/dev/vdb
#      overlayroot=device:dev=/dev/sdb,timeout=180
#      overlayroot=device:dev=LABEL=my-flashdrive,timeout=180
#
#  * overlayroot=crypt:PARAMETERS
#    use an encrypted [dmcrypt] device as the backing device. Parameters
#    are comma delimited key=value pairs.
#
#    available parameters are:
#     * dev: default: "" [REQUIRED]
#       use given device for backing filesystem.
#     * mapname: default: "secure"
#       the name of the map device to be created in /dev/mapper
#     * pass: default: ""
#       if not provided or empty, password is randomly generated
#       the generated password will be stored for recovery in
#       /run/initramfs/overlayroot.passwd
#     * fstype: default: "ext4"
[... 93 lines cut ...]
#    IT IS INSECURE TO SET THIS PASSWORD HERE IN THIS CLEARTEXT CONFIGURATION
#    FILE OR ON THE KERNEL COMMAND LINE.
#    Randomly generated passwords are more secure, but you won't be able to
#    read your encrypted disk on reboot.
#    Randomly generated passwords are generated by calculating the sha512sum
#    of a concatenation of:
#      - stat -L /dev/* /proc/* /sys/*
#        + some unpredictability of access/modify times of a number of kernel
#          files, directories, and block devices
#      - /proc/sys/kernel/random/boot_id
#        + 16-bytes uuid, consider this a 'salt'
#      - /proc/sys/kernel/random/uuid
#        + 16-bytes uuid, consider this psuedo randomness
#      - /dev/urandom
#        + 4096-bytes of psuedo randomness
#      - $DEV
#        + 4096-bytes from the head of the disk
#        + security-paranoid users can write 4096-bytes of randomness to
#          this device and specify mkfs=1 before rebooting into an
#          crypt+overlayroot setup
#    The result is stored in r-------- /dev/.initramfs/overlayroot.XXXXXXX,
#    which is a tmpfs in memory.
overlayroot_cfgdisk="disabled"
overlayroot=""
```

RobertCNelson, 2021-08-16:
PS, since we are tweaking /boot/uEnv.txt  We should define a varible to be used by default..

```

console=ttyS0,115200n8
cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet video=HDMI-A-1:1024x768@60e

#Use an overlayfs on top of a read-only root filesystem:
#cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet overlayroot=tmpfs

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
##cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
```

Should we use:
```
default_cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet
```

and then beagle-config can use that as a reference for tweaking "cmdline"?

SAtacker, 2021-08-19:
> PS, since we are tweaking /boot/uEnv.txt We should define a varible to be used by default..
>
> ```
>
> console=ttyS0,115200n8
> cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 [...]

That would be ideal. Thanks!