SAtacker/bb-config #50: Feature request: Link click for automated bug report

jadonk, 2022-07-15:
It would be great if I could follow the links on the about page. What's more, it would be great if I could get an automated bug report that would pull together information from beagle-version and bb-config.

In particular, I downloaded a recent image and tried to use it to configure the WiFi, but the scan simply never returns anything. I wanted to file a bug report, but I think gathering the right information might be difficult.

SAtacker, 2022-07-16:
Thanks, @jadonk  , the network manager support is in development by Jian De. I agree with you regarding error messages, this should be more verbose.

SAtacker, 2022-07-16:
Let me work on it and get back to you after a few days.